AOSP-JF-MM/issue_tracking_aosp #88: Gallery crashes when refused permission to location.

codeseven58, 2016-09-20:
Like the title says. I don't want anybody to know my location... especially if theyre trying to force me to allow them access.

http://pastebin.com/sb2QTHCC

B--B, 2016-09-20:
Gallery needs the permission for show on map option (geotag), if I remove the permission check gallery opens, then you'll get a security exception when trying to use show on map feature... Don't think it's a good idea remove the check and let gallery open

codeseven58, 2016-09-23:
I hate it when companies "force" you to let them have their way. I ended up reverting to the 08/12 build since it doesn't do that crap. The odd thing is, that Maps was doing the same thing for a while, then after 1 update, it just stopped FCing when I hit deny.

codeseven58, 2016-09-25:
Is there any way to disable gallery's new "show on map" option? I don't want to allow any app to access my location? except for maps for directions.

closed 2016-10-04 by B--B