AOSP-JF-MM/issue_tracking_aosp #90: SIM card not found on samsung galaxy s4 sph-l720 under [JDCTeam][6.0.1][10 September][OMS-Substratum]

kratovonar, 2016-09-30:
Hi, I just flashed your rom to my galaxy s4 sph-l720 and, everything works fine apart from the sim card that says to insert, even though my simcard is placed in my phone, can you tell me which info or log I need to send to you?

codeseven58, 2016-11-14:
2 things, if your phone starts up and it doesnt show the sim card is present, reset the phone. This was a small bug that would show the sim csrd on tue second start. Also, upgrade to the october release and report back with conclusion.

Other thing is if the sim card does work, but randomly disconnects while youre using it, then you have a hardware related issue that can only be fixed by opening your phone and placing a piece of foam over the connector for the sd and simm card then closing up your phone. There aretutorials for this online, google it.

kratovonar, 2016-11-14:
Thank you very much, the october release is much more better and stable I
think, everything works fine now

codeseven58, 2016-11-15:
If you would be so kind as to close this issue, the community and devs would greatly appreciate it. Thank you in advance :)

Suncatcher, 2017-03-05:
I flashed the latest firmware from 22.02 and have exactly the same problem. My SIM-card is not recognized.
Where can I download old builds to try?

smeroni68, 2017-03-05:
Check newly on romhut... now it's visible. Let us know if the old build solve or not. Thanks

Suncatcher, 2017-03-05:
> Let us know if the old build solve or not
Nope, that haven't helped. The same result also on 17/11 build. Maybe worth to try October's release, kratovonar [wrote](https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/90#issuecomment-260280691) about?

Suncatcher, 2017-03-05:
What baseband version should I have for your ROM to work? Now it's L720VPUGPI2

Suncatcher, 2017-03-05:
Flashed the Stock_kernel from you page, but that didn't help anyway.

smeroni68, 2017-03-05:
Flash latest stock rom for your model to check if sim detection is working.

Suncatcher, 2017-03-05:
> Flash latest stock rom for your model to check if sim detection is working.

Perfectly working. I had  L720VPUGPI2_L720SPTGPI2_SPR before and it worked.

smeroni68, 2017-03-05:
Is your phone/network CDMA or GSM?

Suncatcher, 2017-03-05:
> Is your phone/network CDMA or GSM?

GSM. 
BTW, I found out that the SIM is not recognized on **any** ROMs, both stock and custom ones. I installed stock GPI2 via Odin - nope, then tried to install [Exodus](https://forum.xda-developers.com/galaxy-s4-sprint/orig-development/rom-team-exodus-jfltespr-t3135862) - nothing, even my own TWRP backup didn't work all in all!
This thing happened after installing of your jdcteam_jdc_jflte-6-0-1-20170222-stable-zip. Can you explain this?

smeroni68, 2017-03-05:
A pure coincidence.
All custom rom on S4 do not touch the firmware, this is why we write in OP that latest firmware of modem must be installed separately. If you confirm that no one rom (custom and stock) will recognise anymore your sim, i can imagine that the hw sim module do not contact the sim chip anymore. That's my explanation as you requested.

Solution: search around for a new sim reader module (10$) and provide the exchange. Good luck.

Suncatcher, 2017-03-05:
It seems you're right. Will investigate further.

smeroni68, 2017-03-05:
There is another possibility, but dunno how to check: you may have lost IMEI...

rigo9, 2017-03-07:
I am having the same No sim card detection issue on almost all the Roms I have tried so far including this one. Sim only works with stock and that too after selecting gsm in network settings. Any fix for this? My Baseband is L720VPSGPL1.

Suncatcher, 2017-03-07:
> EDIT: go on phone info and check if IMEI is there or is null...

IMEI is showed.

Suncatcher, 2017-03-07:
> I am having the same No sim card detection issue on almost all the Roms I have tried so far including this one. Sim only works with stock and that too after selecting gsm in network settings. Any fix for this? My Baseband is L720VPSGPL1.

Holy shit! You're damn right! It's just should be changed to GSM and that's it! It works on stock!

> search around for a new sim reader module (10$) and provide the exchange

The problem has nothing to do with SIM module. The problem with your implicit LTE-enable on low-lever somewhere in hardware menu.

rredford, 2017-03-10:
I fixed mine by using 2 clear tape covers on top of SIM card. Worked fine most times. I think foam on top would fix it for good also.

B--B, 2017-04-03:
This issue belongs to MM, but next release will be the last, so i close this as MM reached the EOL, N is our stable release now, and unfortunately i have no time to keep two different ROMs alive for multiple devices, BTW with N stable and bug free (wifi hotspot will be fixed in next upcoming release) i don't see the sense to stay with an older android release and OMS working only in legacy mode

closed 2017-04-03 by B--B