AOSP-JF-MM/issue_tracking_aosp #91: Accelerator not working

terraniux19, 2016-10-08:
I have this problem while playing games that use the accelerator or while rotating the screen and it doesn't work. I tried also check the sensor in Sensor Test app and it doesn't show anything. This is Stable7 build for the S4 i9505

B--B, 2016-10-08:
Tried just now with both jflte and mi5 with accelerometer analyzer and it's working fine for me, rotation works too without any issue

terraniux19, 2016-10-08:
@B--B Ok, maybe I'll try the app you mentioned.

terraniux19, 2016-10-08:
I tried the app and it isn't giving me any inofrmation about accelerometer. It seems to be broken.

terraniux19, 2016-10-08:
Ok, that a little weird, but everything came back to normal when I restarted the phone.

smeroni68, 2016-10-08:
Maybe next time you can reboot and check a second time before opening  an issue! Thanks.

closed 2016-10-08 by smeroni68

terraniux19, 2016-10-08:
Ok, now the sensor stopped working after a while. So my conclusion is that the sensor is broken. I think I need to give for repair.