AOSP-JF-MM/issue_tracking_aosp #92: UI issue, low priority, Charging icon

codeseven58, 2016-11-14:
I have the charging icon, in the notification bar, set to circle with charge number inside the circle. The circle used to rotate while it is charging but no longer does that. However when I frag down the notification bar, the icon then acts normally and rotates.

dkati, 2017-03-17:
mi5 or jf mister :) ?

B--B, 2017-04-03:
This issue belongs to MM, but next release will be the last, so i close this as MM reached the EOL, N is our stable release now, and unfortunately i have no time to keep two different ROMs alive for multiple devices, BTW with N stable and bug free (wifi hotspot will be fixed in next upcoming release) i don't see the sense to stay with an older android release and OMS working only in legacy mode

closed 2017-04-03 by B--B