AOSP-JF-MM/issue_tracking_aosp #94: Offline Charging Broken

style2345, 2017-03-16:
When Device is in Switch Off Condion & Connect to Charger then it Shows ( ??/100 )

B--B, 2017-03-16:
Thanks for the report
Fixed by this https://github.com/AOSP-JF-MM/platform_device_xiaomi_gemini/commit/61866708957ddb57c50c15fd3d66cabab4d8c6d5 and this https://github.com/AOSP-JF-MM/platform_system_core/commit/92d71cb83cac5edd18561808c0f40c2f2ef06f11
BTW charging function works in public release, which is the most important thing, only the charging images were missing
Regarding battery percentage it's a caf change, i've merged that too, tested now and it' s working fine
Please provide us some logs for the other bug you opened as i don't have any disconnection issues and i'm not able to reproduce your problem

closed 2017-03-16 by B--B

style2345, 2017-03-17:
Rebuild the Rom With this Charging Image Patch Or Provide OTA Update 

Thanks ..

smeroni68, 2017-03-17:
Internal build already running... PM me on xda account...

style2345, 2017-04-03:
This issue has been fixed in

ROM Version = [JDCTeam][7.1.1][28 March][Substratum][EAS] - Stable 3

Thanks