AOSP-JF-MM/issue_tracking_aosp #102: Can´t update

ibiza69, 2017-04-13:
Hi, thanks for the ROM 😉

I got an update for your rom, I had 7.1.1 and you got an update for 7.1.2 at the mobile update section. I proceeded with update, but after downloading and restarting from recovery it said couldn´t update: Can´t open /data/media/0/OTAUpdates/JDCTeam_jdc_jflte-7.1.2-20170408-stable.zip [bad] installation aborted…

So, I restarted again the phone in TWR recovery mode and tryed to update my self manually the zip file, from SD, but got same error.

Can you please give a hand on what could be happening, why can´t I update (Automatically or manually) from 7.1 to 7.2, it´s same S4 LTE AOSP rom, etc…

Thanks!

smeroni68, 2017-04-13:
Download from a pc the rom. OTAUpdates has a problem and get wrong file. Thanks.

ibiza69, 2017-04-13:
Thanks, updated as you said ;) Downloaded to a PC, then to the phone and then from recovery ;) You are the best, it´s incredible to have our samsung S4 working with Nougat! Great job and thanks for your dedication :)

B--B, 2017-05-01:
FIxed in our latest public release. this can be closed

closed 2017-05-01 by B--B