AOSP-JF-MM/issue_tracking_aosp #104: Not able to send/receive SMS

romaniscu, 2017-04-18:
After installing 7.1.2 (April 8th update), my phone is not able to send or receive SMS. When trying to create new SMS, and after choosing recipient, the Messaging app doesn't offer me any way to compose and send a message. I've installed Google Messages app but but after choosing recipient, the app displays the following/ Failed to create conversation. Please try again letter.

icemen, 2017-04-23:
solution: [https://forum.xda-developers.com/showpost.php?p=71886256&postcount=2294](https://forum.xda-developers.com/showpost.php?p=71886256&postcount=2294)

smeroni68, 2017-04-23:
Just to write here once for all:

Google has changed the structure of sms database... do a clean flash.
Or in alternative follow this procedure:

Reboot in recovery, **delete** the folder **/data/user_de/0/com.android.providers.telephony** and reboot in system, this will recreate the sms/mms database.

Do a backup of sms on previous build before do the rebuild. After you can restore the messages.

closed 2017-04-23 by smeroni68