AOSP-JF-MM/issue_tracking_aosp #106: no call audio through wired headset

markun, 2017-05-07:
When I try to make a call or pick up a call while my (wired) headset is connected, I don't hear the audio from the call and the other side does not hear me.

When I hangup the call, I do hear the "beep" that marks the end of the call, just the audio from the phone call seems to be muted for some reason.

I just did some experiments by calling my voicemail:

1. I call with my headset connected and I hear nothing.
2. When I press the speaker phone button (while the headset is connected) I hear nothing
3. When I disconnect the headset and listen to the phone I still hear nothing
5. Now, when I press speaker phone (headset still disconnected) I hear the call!
6. when I take the phone off speaker phone I still hear the call
7. when I connect the headset, I do hear the call through my headset!

btw, when I go directly from 5 to 7 I don't hear the call

After I go through the cycle above (headset, unplug, speaker phone, off speaker phone, headset) I am able to make a few normal phone calls using either my headset or just using the handset.

What are the ways to debug these kinds of audio issues? What kind of logs do you want me to provide?

smeroni68, 2017-05-07:
You do not reported the model of your phone.
This seems to me a missing contact in the audio jack (hw problem). The headset is detected as inserted, and the audio redirection through the jack is done. But probably the contact inside the jack block the audio.
Do you have the possibility to test another rom to understand if this is the problem?

markun, 2017-05-07:
The model is i9505.

In my original message I wrote that I can get audio during calls with my headset to work by unplugging the headset, toggling speaker phone on and off and inserting the headset again. Audio playback from my music payer (and even the hangup beep after the call) always work fine through the headset.

@smeroni68, can you explain how the hardware issue you talk about can cause the described behavior?

I recently came from the last official rom from Samsung that was available for my phone. I did a clean install of AOSP, wiped everything, but since then I have this issue.

Is there some adb tool that allows me to see how audio is being routed?

smeroni68, 2017-05-07:
To have informations, you need to capture the logs right after the issue appear. So follow the sequence you described and capture the logs (eg with syslog app) and provide here a link to them.

As you know, we moved to MI5 so S4 isn't anymore our dayly phone.

markun, 2017-05-07:
Just to be clear: capture logs after I make a phone call with headset connected or after I do the trick to get audio working again?

smeroni68, 2017-05-07:
Log are persistent for some time... Do the complete testing as in your list from 1 to 6 actions than capture all logs and share them here (upload them to a share and put here the link).

markun, 2017-05-07:
[logs after steps 1 to 6](https://marcoen.tk/SysLog/2017-05-07_14.22-after-trick-to-get-audio-working.zip)
And just in case:
[logs after no call audio and hangup](https://marcoen.tk/SysLog/2017-05-07_14.21-no-call-audio-with-headset.zip)

markun, 2017-05-12:
@smeroni68 did you have a chance to look at the logs? Anything suspicious in there?

fratre-eiti, 2017-11-02:
@smeroni68 same identical issue with my S4. After 3 (always three) plug/unplug of the jack the earphones start to work properly.

Francesco

smeroni68, 2017-11-02:
Hi Francesco,
I restored AOSP N from my last backup, and I tested the wired headset (original Samsung for S4)... I'm sorry, but here it's working normally from first plug-in. I tryed 7 times, and all times was working perfectly. 
I tested also Oreo and all works also there. Sincerely I dunno what to answer you.
I can only say that on over 10.2K users that has downloaded latest public version and used it, only you and few others has problems with the headset... My thinking about this issue I've already posted on xda... you have probably a miscontact on the jack or in the socket that prevent the right switching from speaker to headset (the switch is by software, but the trigger for it is electro-mechanical).
If I'm right, maybe that searching for a spare socket to be exchanged (the cost itself is few euros) can solve your problem, but it's just a suggestion and you are free to follow or deny.

Dunno how to help you, sorry.

fratre-eiti, 2017-11-03:
Hi @smeroni68. Thank you very much for writing back to me. The only thing I can try is to use the original headset. With this phone I always used a Sony headset.. Maybe this is the reason. 

One more question just to be sure.. Tha latest version of this ROM is the one linked in the first post at https://forum.xda-developers.com/galaxy-s4/i9505-orig-develop/jdcteam-android-source-project-nrd90m-t3453084 (https://romhut.com/roms/aosp-jf-7-0)  ? #### STABLE 9 (20170821)

If it neither worked with Samsung original headset I'll try to clean flash the phone again.

fratre-eiti, 2017-11-14:
@smeroni68 I finally solved by reflashing my S4. Clean install, as I did the first time however. Now hedsets, of any brand, work fine. Thank you ;)

smeroni68, 2017-11-14:
Glad you solved! I'll close this issue.

closed 2017-11-14 by smeroni68

fratre-eiti, 2017-11-16:
@smeroni68 I'm sorry to inform you that the issue has come back again. One day after the reflashing. Original headsets or not I have to replug the jack 2 or three times for the microphone and earphones to work properly. If I can help with logs I'll willingly collect them.. Just tell me how to do it ;)

PS: I also cleaned the jack socket with a electro-reactivator with no luck. I'm pretty sure that it's not an electric issue. I used it without problems all day long yesterday.

Thank you,
Francesco

smeroni68, 2017-11-16:
Dunno mate, we are not using anymore the S4, and is not a general issue. Also the test phone of BB is nearly arrived to eol... so really do not know.
It can be a sw problem, but we have no time to investigate for it and only for few users. Sorry.