AOSP-JF-MM/issue_tracking_aosp #107: Build - Add devices specific repos dependencies

B--B, 2017-05-16:
I'm writing here so everyone can read, we should really implement device specific repos dependencies, we added new devices to build barrier (and hopefully we will add more devices soon), this will add more and more repos to our build enviroment, and will require everytime more space on our ssd for unuseful repos, and more time for repo sync. Honestly i never cared about this until now because we were bulding only for a few devices, so i don't know how other ROMs implemented this function :) if someone already took a look at it feel free to merge everything in our repos, otherwise i will take a look hopefully this weekend, sorry but real life comes first :)

smeroni68, 2017-05-16:
A way to solve has been already pushed from MZO:

https://github.com/AOSP-JF-MM/platform_manifest/commit/587a48d1a0f6e432ca5cdabb9123950a02303aaf#commitcomment-22130344

Need to check if it's fine.

xlxfoxxlx, 2017-05-20:
roomservice could be implemented, you just have a dependencies file in device tree and then when you lunch the devices if fetches only needed device repos.

smeroni68, 2017-05-20:
Welcome Fox.

B--B, 2017-05-27:
Tested a working on Mi5 and Jf, other devices may need some small change but roomservice is working pretty fine. Additional repos can be added to dependencies (like device specific HALs) but we can move them later.

closed 2017-05-27 by B--B