AOSP-JF-MM/issue_tracking_aosp #108: Android password protection

Pietia10, 2017-05-26:
Hi,
There is a problem with password protction in ROM , (try submitting bad password five times)it simply does not work at all even though there are sources in aosp for that and it should be working .

smeroni68, 2017-05-26:
How is meant to work?

Pietia10, 2017-05-26:
The code is in aosp KeyguardSecurityContainer.java

smeroni68, 2017-05-26:
@B--B @MZO9400 @CheckYourScreen @xlxfoxxlx ... something to check here.

smeroni68, 2017-05-26:
@Pietia10 This is the code... it is disabled by default:
https://github.com/JDCTeam/android_frameworks_base/commit/13cb572a1a54f12fec1b82bcbe0254e2f4fcf120

Pietia10, 2017-05-26:
Yeah in every phone this is enabled by default.

smeroni68, 2017-05-26:
Looking on the code, if we set eg. 5 times, it will ask to factory reset the data... so you would like to have this?

MZO9400, 2017-05-26:
True, doesn't work

smeroni68, 2017-05-26:
It is meant to be so if disabled... as it is.

MZO9400, 2017-05-26:
At least it should timeout. It doesn't timeout after 5 attempts

smeroni68, 2017-05-26:
+                if  (!(mMaxCountdownTimes > 0) && timeoutMs > 0) {
                      long deadline = mLockPatternUtils.setLockoutAttemptDeadline(		                      long deadline = mLockPatternUtils.setLockoutAttemptDeadline(
                              userId, timeoutMs);		                              userId, timeoutMs);
                      handleAttemptLockout(deadline);		                      handleAttemptLockout(deadline);
                  }		                  }
              }		              }
              if (timeoutMs == 0) {		              if (timeoutMs == 0) {
 -                mSecurityMessageDisplay.setMessage(getWrongPasswordStringId(), true);		 +                String msg = getMessageWithCount(getWrongPasswordStringId());

MZO9400, 2017-05-26:
Weird

Pietia10, 2017-05-26:
If it wont factory reset at some point the password will be easily bruteforceable , google know their's stuff .

smeroni68, 2017-05-26:
My mistake... the commit I linked is on Opt-CM14 github...

MZO9400, 2017-05-26:
Ah lol

smeroni68, 2017-05-26:
Can we think to change this behaviour? Finally I use 99.99% the FP unlock...

smeroni68, 2017-05-27:
FYI: if you fail for 5 times the FP recognition... the message that comes out is "Too much trials: retry later"... and you need to use the pin.

Pietia10, 2017-05-27:
Not true with fp you just wait 30 seconda and try again .

smeroni68, 2017-05-27:
What is not true?

Pietia10, 2017-05-27:
You dont need to use pin after 5 failures of fp you just wait 30 seconds
and you can use fp scanner again.

smeroni68, 2017-05-27:
If you need to unlock the phone, you will wait 30 seconds, or you will swipe the correct pin to use the phone?

Anyway, the behaviour (as already  written) is the same on LOS and all custom roms based on it. That's the situation.

Pietia10, 2017-05-27:
The fingerprint implementation is same as in aosp however unlocking with
passcode is not .

smeroni68, 2017-05-27:
Are you referring to swipe passcode (3x3 or 4x4) or numeric keyboard with pin?

Pietia10, 2017-05-28:
Im referring to both.

smeroni68, 2017-05-28:
But numeric keyboard pin works!

[https://www.dropbox.com/s/8qo0ltqznzl1xjx/Screenshot_20170528-172633.png?dl=0](url)

Only swipe sequence has no timeout. Please check and confirm.

Pietia10, 2017-05-28:
I cant reproduce what you sent me it doesnt work on my mi5 17 may build.

smeroni68, 2017-05-28:
Fine... on our internal build, after 5 wrong pins or fingerprint, start the timeout of 30 seconds... on sequence swipe not.
We will check what's happen... for next public build.

Thanks for reporting.

smeroni68, 2017-05-28:
I found something: after a reboot, if you fail 5 times the sequence it stop with 30 sec delay.
After entering the good sequence, the failing of 5 times do not generate anymore the 30 sec delay.
To be checked...

Pietia10, 2017-06-07:
Isnt a good idea to copy keyguard implementation from plain AOSP?

smeroni68, 2017-06-20:
The issue do not exist, security it's fine. Closed.

closed 2017-06-20 by smeroni68