AOSP-JF-MM/issue_tracking_aosp #110: 2 bugs 1 suggestion

PR1D3S, 2017-07-04:
1st bug is when i click on wallpaper to change wallpaper on inbuilt launcher getting error launcher 3 keep stopping...
2nd bug is when i move any file to another folder getting error "couldn't move 1 file even file moved...
3rd is suggestion that inbuilt system icon should be pure and stock like lineage OS because round icon doesn't fit with another icons...

samithagun, 2017-07-05:
I was able to fix the 1st bug by installing the Google Now launcher. When
it has installed, wallpapers section will display correctly.

B--B, 2017-07-12:
1: you need google wallpaper apk, and it's not a bug, but stock AOSP code, BTW after Launcher3 updates this does not happen anymore
2: already fixed here: https://github.com/AOSP-JF-MM/platform_packages_providers_MediaProvider/commit/7555a8aac2f80c68d49352a2044bacbbffcec6b3 and here: https://github.com/AOSP-JF-MM/platform_packages_providers_MediaProvider/commit/5681c7060272fc43eb46fb5c59dfceb6be5b550e
3: the goal is to have ALL round icons like stock Google ROMs, so i don't think i will go back to old icons, but now Laucher3 supports icon packs, so it's not a big issue, just install your favourite icon pack and apply it :)

closed 2017-07-12 by B--B