AOSP-JF-MM/issue_tracking_aosp #111: severe battery drain

pnutts, 2017-07-06:
with latest update ( updated July 2 /17 ROM 20170624 ) I have notice severe battery drain when on standby with S4 i337M phone running nougat 7.1.2. Never happened before.

smeroni68, 2017-07-06:
After two weeks after releasing? When you installed it?

pnutts, 2017-07-06:
I installed it on July 2/17 thru OTA update.
I have reverted back to earlier build for now.
Will wait until a newer build is released.

smeroni68, 2017-07-06:
Rom has no battery drain itself. It's something related to your installation. For your information, user that download the build are 4068 right now... and we do not see on xda the thread full of posts about severe battery drain.
Wait for next build, but as wrote, is not rom dependent.

pnutts, 2017-07-07:
I still will wait for next build and do it manually - not OTA update.
No battery drain problem with previous build since re-install.

B--B, 2017-07-22:
No one reported battery drain after our latest public release, so I assume it's fixed

closed 2017-07-22 by B--B