AOSP-JF-MM/issue_tracking_aosp #112: 7.1.2 AOSP NJH47B - GPS Issues

RuKaDa, 2017-07-10:
Performed a clean install, latest modem and bootloader is installed. Everything is working fine with this ROM (BTW its awesome, thanks to developers). The only issue is regarding with the GPS, it can only fix if the airplane mode is turned ON. When turned OFF i can´t get any kind of fix, tried with all the possible combinations that i´ve remember Wifi, mobile data, BT, only GPS, high accuracy and so on. Nothing seems to get a fix. Please find the attached log, hope that helps to find it.

Best Regards to all

[2017-07-10-20-57-08.zip](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/1136951/2017-07-10-20-57-08.zip)

smeroni68, 2017-07-10:
When was last time you get a fix with the rom?

RuKaDa, 2017-07-10:
About 15 minutes before my previous post, with airplane mode turned ON, if i set it to OFF the GPS fix is like a beacon, on for a few seconds and off, and it keeps this way.

smeroni68, 2017-07-10:
Did you use any app to check the GPS position?

RuKaDa, 2017-07-10:
I´ve tried GPS status & Toolbox, Faster GPS, both have found at least 15 satellites, although when using navigation app (tried Sygic, iGO and Meo Drive) i cant get a fix. Can i ask you if you find something wrong in the log file?

smeroni68, 2017-07-10:
I haven't checked yet... but in the mean time I whent out of my house and switched on the GPS and app (Status GPS)... it took around 20 seconds:

https://www.dropbox.com/s/gnqlvdcr9erfyv1/Screenshot_20170710-223645_clear.png?dl=0

No tricks, just switched on as High accuracy and updated agps data...

RuKaDa, 2017-07-10:
With Status GPS i´ve also had a fix, really fast fix btw, but with navigation apps it doesn´t work. I´ll try it again with Status GPS, with GPS only, i don´t have much mobile data, sorry for that. Back in a few minutes.

smeroni68, 2017-07-10:
You wrote you installed latest FW... version?

RuKaDa, 2017-07-10:
Latest Stock ROM from updato, rooted with chainfire root, installed twrp and after a few days digging around XDA found this and install it

smeroni68, 2017-07-10:
So you do not know exactly the fw version?

smeroni68, 2017-07-10:
You can get from here

https://forum.xda-developers.com/mi-5/development/tool-flashable-firmware-twrp-easy-t3474898

the version 7.5.4 and flash in twrp.

smeroni68, 2017-07-10:
Just a question: have you wiped the phone completely before installing custom roms? Or let's say better: have you formatted the internal sd (loosing everything on the phone)?

smeroni68, 2017-07-10:
I'm having a look into the log. But I see some errors app related:

07-10 20:54:07.776 E/navngo  (17725): Config:find_libEGL:egl.cfg not found
07-10 20:54:07.776 E/navngo  (17725): java.io.FileNotFoundException: /system/lib/egl/egl.cfg (No such file or directory)
07-10 20:54:07.776 E/navngo  (17725): 	at java.io.FileInputStream.open(Native Method)
07-10 20:54:07.776 E/navngo  (17725): 	at java.io.FileInputStream.<init>(FileInputStream.java:146)
07-10 20:54:07.776 E/navngo  (17725): 	at java.io.FileInputStream.<init>(FileInputStream.java:99)
07-10 20:54:07.776 E/navngo  (17725): 	at com.navngo.igo.javaclient.utils.StreamUtility.readFile(Unknown Source)
07-10 20:54:07.776 E/navngo  (17725): 	at com.navngo.igo.javaclient.Config.find_libEGL(Unknown Source)
07-10 20:54:07.776 E/navngo  (17725): 	at com.navngo.igo.javaclient.NNG.setParams(Unknown Source)
07-10 20:54:07.776 E/navngo  (17725): 	at com.navngo.igo.javaclient.NNG.reInit(Unknown Source)
07-10 20:54:07.776 E/navngo  (17725): 	at com.navngo.igo.javaclient.NNG.<clinit>(Unknown Source)
07-10 20:54:07.776 E/navngo  (17725): 	at com.navngo.igo.javaclient.IgoActivity$5.run(Unknown Source)
07-10 20:54:07.776 E/navngo  (17725): Config:find_libEGL: better then nothing: /system/lib/egl/libGLES_android.so

Seems the app is searching some graphic libraries in the wrong place... not sure, but seems so.

RuKaDa, 2017-07-10:
Yes, fully wiped, without sdcard, just went outside and got a fix of 4 of 20, but with the beacon style again. Installed this one: I9505XXUHPK2_I9505TPHHPL1_I9505XXUHPK2_HOME.tar.md5, then followed instructions for install AOSP 7.1.2

smeroni68, 2017-07-10:
Oops... you are on S4 I9505... not on MI5... forgot all stuff about firmware!!

RuKaDa, 2017-07-10:
yes i noticed that, I9505 it´s my phone, but that´s fine :). So i believe that i made it the right way, starting point: I9505XXUHPK2_I9505TPHHPL1_I9505XXUHPK2_HOME.tar.md5, then rooted with chainfire and installed twrp. After that fully wiped storage (without sdcard) and installed AOSP

smeroni68, 2017-07-10:
Yes. But is iGo app starting and working? From the log i see a lot of errors on graphic library side...

RuKaDa, 2017-07-10:
Yes, all the navigation apps that i´ve tried are working and they get a fix but after a few seconds they loose it.

smeroni68, 2017-07-10:
Have you restored the app from a titanium backup done on previous stock rom? (or something similar)

smeroni68, 2017-07-10:
Probably ( i can assume) you are having a loosing of signal from antenna contacts... it's a known issue on S4...

RuKaDa, 2017-07-10:
It could be, but with stock rom it doesn´t happen. No restored app, made a fresh and clean install of all apps that i´ve tried. There´s also a user who reported issues with gps with stable 5 but i´m on stable 7 if that could help.

RuKaDa, 2017-07-10:
Let me go outside again, just to check if i get a fix with status gps, back in 2 minutes

smeroni68, 2017-07-10:
Sorry, but looking on the log it seems the app itself has a lot of errors on library... and sincerely, due to the fact that S4 with the "age" is giving also this kind of faults... If you wanna be sure, reinstall stock rom and test. Can't help more about this matter.

RuKaDa, 2017-07-10:
Ok, thanks for your help, just got a 5 out of 18 fix with clear sky, GPS only mode, with airplane mode OFF and ON, do you think that the SIM CARD could eventually be a problem?!

smeroni68, 2017-07-10:
Really never heard about the sim card can generate this kind of instability.

RuKaDa, 2017-07-10:
I´ve read about it, because of the APN settings for data. Anyway maybe i´ll try with another AOSP ROM, maybe MM, can you point me one as clean as this?!

smeroni68, 2017-07-10:
On S4 I know only JDCTeam roms... but you must reinstall stock rom to test. If I'm right, you will have similar problems. Defects do not advise... they happens... and you must know.

closed 2017-07-10 by smeroni68

dkati, 2017-07-10:
Please try stock kernel and report back 

Alucard kernel seems to have a strange issue with qmuxd. Happens on optcm sometimes too

RuKaDa, 2017-07-11:
Hi to all,

Just to give some feedback about the GPS issue in this ROM...

Flashed Official Stock ROM LL 5.0.1 from Sammobile, GPS issues are gone, got a fix of 10 satellites in about 12 seconds, in a cold start it takes 20 seconds to get a fix, performed all these tests with GPS only mode, no wifi and no mobile data, with Airplane mode turned ON and OFF. Also got a 5 satellites fix indoor... Outdoors, the minimum that i´ve got was 6.

With Stable 7 i only get a fix with Airplane mode ON, so i assume that isn´t an hardware issue with my device, and there´s no problem with the GPS antenna also, there´s something that isn´t working with this Stable 7, at least for me, and without being able to use GPS only it doesn´t fit my needs... Too bad because imho this is still an amazing ROM, and despite all the things that didn´t work out well for me, congrats to the developers, testers, and all that somehow where involved on this...

Brgds to all

B--B, 2017-07-12:
I' ve tried now with both my jf devices and GPS lock works fine, tested with/without GAPPS/SIM/Airplane mode, i remember that some people experienced similar GPS issues in different custom ROMs, for a lot of them the fix was to flash a stock ROM, do a GPS fix and then flash again the custom ROM, even if honestly i don't know why this fixes the GPS lock, and why sometimes it doesn't work properly after a custom ROM flash. We should release soon the new build, if you have time please flash it and let me know if GPS works, if it's still broken please provide me logcat, dmesg and radio logs with the GPS fix in Airplane Mode ON and OFF so i can take a look at it :)

@dkati you should take a look at your Ramdisk if you experience strange issues, as no one reported problems related to QMUXD here , if for some reason QMUXD crashes you should have an unknown baseband and RIL stop working, this is not the case for what i read

RuKaDa, 2017-07-13:
@B--B Thanks for spending time for testing and for your reply, when released i´ll give it a try for sure, let me ask you if the modem and bootloader versions can make the difference or can they somehow related to my GPS issues? Currently my modem and bootloader version is  XXUHQC1, i believe is the latest available in XDA.

RuKaDa, 2017-07-13:
@B--B and @smeroni68 

I think that i might have found a temporary solution for my GPS issues, on my way work this morning i've started my navigation app, no GPS fix again. As i was jammed in the traffic i've made a s[U]oft restart with location on device only turned ON[/U], after the restart got a GPS fix almost instantaneously, and my navigation app "drive me" to work, during this morning i've made some tests:

If i disconnect the location after a while, there's no GPS fix again.
Soft restart with location on device only ON and GPS fix in a few seconds. :good:
Despite this is not a permanent fix, this solution for me is acceptable, and i hope it works every time i need it and also works for those who having GPS issues :D

Perhaps this is related to power management, i will perform some more tests with the performance profile and automatic optimization changes, just to check if the system keeps the location on.

Thanks for your patience and for trying to help me to get this working, best regards.

dkati, 2017-07-14:
@B--B there are some situations where qmuxd never shuts down even on airplane mode.i reproduced it while the logcat was disabled via STweaks.never understood why that is being happent. 
talking with mattia,he confirmed it and said its an old bug of alu kernel

B--B, 2017-07-14:
@dkati QMUXD should never stop, even in airplane mode, no matter what kernel you use. 

https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/blob/aosp-7.1.2/rootdir/etc/init.qcom.rc#L446

It starts here like all other main class services and never stops

You can see by yourself checking logcat and radio log with airplane mode on