AOSP-JF-MM/issue_tracking_aosp #115: Bluetooth Share Stoped

Zhaarr, 2017-07-22:
A pop-up message keeps appearing

"Bluetooth Share Has Stopped"

Even if Bluetooth is off

smeroni68, 2017-07-22:
When you report an issue, can you please detail what are you using? We do not have a chrystal ball to know what you have inyour hands!

Phone version
Rom version and date
Baseband version
Gapps package and date

Zhaarr, 2017-07-23:
Phone: Samsun Galaxy S4 GT I9505
Rom : ASOP 7.1.2 Version 20170716
Baseband Veresion: I9505XXUHPK2
Gapps : Arm 7.1 Aroma

The Bluetooth does not even come one at this point...

My bad for not giving details, I assumed that since I got here by clicking a link from Galaxy S4 form, that this was solely about the Galaxy S4.

Hope I can get my Bluetooth working soon.

Regards

smeroni68, 2017-07-23:
Aosp issue tracker is shared between devices, so always are needed details. We support S4, MI5, OneplusX/2/3/3T.
Regarding the problem you found, is probably related to gapps aroma. 

Please, download the micro package and do a clean install to test.

Zhaarr, 2017-07-23:
Hi,

I downloaded the micro package and did a clean install. But I still get the Bluetooth problem.

smeroni68, 2017-07-24:
We can only tell you to do as this user has done:

https://forum.xda-developers.com/showpost.php?p=73140581&postcount=2695

Zhaarr, 2017-07-24:
Should I just reflash the Rom then, without any wipe...

smeroni68, 2017-07-24:
Yes, I already wrote what you must try...

Zhaarr, 2017-07-25:
Hi,

I did as you suggested, dirty flashed Rom.  Bluetooth is now working but the microphone does not work for skype. Should I open a new thread for that?

Thanks!

smeroni68, 2017-07-25:
First, let's close this issue, than you can open another one... but if you don't get logs and link in the issue, is pretty unuseful and devs can't evaluate the problem... and the issue will be closed.

closed 2017-07-25 by smeroni68