AOSP-JF-MM/issue_tracking_aosp #116: Mic no working during Skype and FB Messenger calls

Zhaarr, 2017-07-25:
Hi am using:

Phone: Samsun Galaxy S4 GT I9505
Rom : ASOP 7.1.2 Version 20170716
Baseband Version: I9505XXUHPK2
Gapps : Arm 7.1 Aroma

I had "Bluetooth Share Stoped" problems as well as the problem with the mic, but the Bluetooth problem was resolved by dirty flashing the Rom.

How ever the mic problem still persists. I tried to dirty flash Rom again to fix this but it was of no avail.

Skype was working with older version....

Let me know if there is any other info required in order to help resolve this issue.

Regards

smeroni68, 2017-07-25:
Already told you to post logs get during the issue... writing here just I have a problem, is not useful... did you get the point?

Zhaarr, 2017-07-25:
What do I need to do? I am not familiar with this process. Please do guide me.

Thanks for your patience.

smeroni68, 2017-07-25:
Sorry, but you are using this channel as xda forum! 
If you are not familiar with an issue report, why coming here? Learn how to get logcat with any tool or app and give developers something they can analyze to understand and maybe fix. Thanks.

smeroni68, 2017-07-25:
If you type on Google search "how to get logcat" you will find many tutorials to lear about.

https://www.google.it/search?q=how+to+get+logcat&oq=how+to+get+logcat&aqs=chrome..69i57j0l5.4844j0j8&sourceid=chrome&ie=UTF-8

B--B, 2017-07-25:
Sorry but this is a bug tracker not xda, I receive a notification every single message, this is not the place to talk about how pick logs, use xda for this kind of questions/issues

closed 2017-07-25 by B--B

Zhaarr, 2017-07-25:
Hi,

Thank you guys for your patience, I came here directly because there is a link in XDA for the ROM to report any issues. So I opened the link to do exactly that. 

I do apologise that I am not prepared to report the problem with sufficient details, I will look into how to get the required details and post it here asap.

Again thanks for your time, patience and understanding!

Regards