AOSP-JF-MM/issue_tracking_aosp #117: Impossible to exit screen pinning

andyrichardson, 2017-08-09:
**Phone Model:** Xiaomi Mi5
**ROM Version:** 20170716
**Clean Install:** Yes
**System Modification:** No
**Tried reboot:** Yes

--

**Reproduction:** Enable screen pinning in settings. Pin any app and attempt to exit by holding back and switcher hardware buttons.

**Expected Result:** App leaves screen pinning.

**Actual Result:** Notification displays prompting the user to press back and app switch to exit.

**Additional Information:** Likely caused by having long press mapped to an additional function (menu) in hardware button settings. 

Other ROMS haven't caused this issue despite having long press mapped to an additional function (should be fixable).

smeroni68, 2017-08-09:
Hi, probably the mapping of the long press has this behaviour... can you try to press the back button for 4 times fastly and see if you will be back to home screen?

B--B, 2017-08-09:
Nice catch! Thanks for the report, it's working fine now :)

closed 2017-08-09 by B--B