AOSP-JF-MM/issue_tracking_aosp #119: [DISCUSSION] JFLTE bringup

B--B, 2017-09-09:
@dkati @OrdenKrieger I open this so there is a place for jf discussion,  actually the features broken here are bt/nfc, camera (we miss all changes needed for legacy devices so it's normal) and ril, all the rest works fine for me, do you guys have public sources? Just for understand differences between my branch and yours

dkati, 2017-09-09:
@B--B personally me ,i didnt push anything yet...still stuck to bootanimation with some ramdisk edits kanged from you...with pure lineage sources i keep rebooting to download/recovery or i dont even pass the splash screen.i already sent u the kmsg

i beleive we need that guy too @itsonlybinary

B--B, 2017-09-09:
Are you using Los sources without changes? You miss core/selinux and other not pushed changes that breaks the boot, you also need text relocations enabled or device won't boot

dkati, 2017-09-09:
i am messing with system/core to fix the init but still....what do u mean with selinux ?Is there a specific path/project to look in ? 

about text relocations,i have enabled em

OrdenKrieger, 2017-09-09:
Yea im actually using Los source in order to make things easier with repopicking etc. I where able to boot the rom 2 days ago with display, bt and wlan working but since yesterday it's broken again. The problem is that I'm stucked at the vendor logo and doesn't even get to the bootanimation so I'm no able to debug anything.

OrdenKrieger, 2017-09-09:
Thats by the way the script from AdrianDC that I'm using to pick common stuff from Lineages gerrit https://github.com/AdrianDC/lineage_development_sony8960_master/blob/local_manifests/repopick-lineage-15.0.md

OrdenKrieger, 2017-09-09:
The only commit that he removed was this: https://review.lineageos.org/#/c/186740/ But I don't think that should effect that at all. Anyways I'm running a new build with that. Im depressed....

dkati, 2017-09-09:
surely not affects @OrdenKrieger ... do u use the external/selinux ?

OrdenKrieger, 2017-09-09:
@dkati Yea just normal LOS one. https://github.com/LineageOS/android_external_selinux/commits/lineage-15.0

dkati, 2017-09-09:
i dont know..thats the latest kmsg i got : 
`
init: kill(1461, 9) failed: No such process
init: service ril-daemon does not have a SELinux domain defined
init: property_set("ro.boottime.ril-daemon", "71866722057") failed: property already set
init: service qcamerasvr does not have a SELinux domain defined
init: property_set("ro.boottime.qcamerasvr", "72017723049") failed: property already set
init: kill(1477, 9) failed: No such process
init: service ril-daemon does not have a SELinux domain defined
init: property_set("ro.boottime.ril-daemon", "76872307275") failed: property already set
init: service qcamerasvr does not have a SELinux domain defined
init: property_set("ro.boottime.qcamerasvr", "77023186197") failed: property already set
init: kill(1485, 9) failed: No such process
init: kill(1489, 9) failed: No such process
init: servIce ril-daemon does not have a SELinux domain defined
init: property_set("ro.boottime.ril-daemon", "81878441809") failed: property already set
init: kill(1501, 9) failed: No such process
init: service qcamerasvr does not have a SELinux domain defined
init: property_set("ro.boottime.qcamerasvr", "82028801931") failed: ppoperty already set
init: kill(1509, 9) failed: No such process
i.it: service qcamerasvr does not have a SELinux domain defined
init: property_set("ro.boottime.Qcamerasvr", "87127435256") failed: property already set
init: service ril-daemon does not have a SELinux domain defined
init: property_set("ro.boottime.ril-daemon", "87890466338") failed: pro0erty already set
init: kill(1517, 9) failed: No such process
inid: kill(1521, 9) failed: No such process
init: service qcamerasvr does not have a SELinux domain defined
init: property_set("ro.boottime.qcamerasvr", "92398859124") failed: property already set
init: service ril,daemon does not have a SELinux domain defined
set_therm_mitigation_Leel: called before init
init: property_set("ro.boottime.ril-daemon", "93790582895") failed: property already set
init: service qcamerasvr does not have a SELinux domain defined
init: property_set("ro.boottime.qcamerasvr", "97404474860") failed: property already set
init: service ril-daemon does not have a SELinux domain defined
init: property_set("ro.boottime.ril-daemon", "99169489684") failed: property already set

5:43

```

init: kill(1235, 9) failed: No such process

init: Write_file: Unable to open '/sys/android_power/request_state': No such file or directory

init: write_file: Qnable to write to '/sys/power/state': Invalid argument

init: killProcessGroup(0, 1234, 9) failed: Device or reskurce busy

init: killProcessGroup(1041, 1243, 9) failed: No such file or directory

init: kill(1243, 9) failed: No such process

init: do_restart: Service audio-hal-2-0 not bound

init: kilLProcessGrou0(1047, 1244, 9) failed: No such file or directory

init: kill(1244, 9) fai,ed: No such process

init: killProcessGroup(1010, 1245, 9) failed: Device Or resource busy

init: jilL(1245, 9) failed: No such process

init: +illProcessGroup(1013, 1248, ) failed: Device or r%source busy

init: kill(1248, 9) failed: No such process

init: property_set("ro.bootdime.netd", "51648122618") failed: property already set

init: property_set("ro.boottime.zygote", "51657583067") failed: property already set

init: couldn't write 1329 to /dev/cpuset/foreground/tasks: No such file or directory

init: service ril-daemon does not have a SELinux domain defined

init: property_set("ro.boottime.ril-daemon", "51842702717") failed: property already set

init: property_set("ro.boottime.audioserver", "51852590412") failed: property already set

init: couldn't write 1337 to /dev/cpuset/foreground/tasks: No such file or directory

init: property_set("ro.boottime.cameraserver", "51872243733") failed: property already set

init: couldn't write 1338 to /dev/cpuset/camera-daemon/tasks: No such file or directory

init: property_set("ro.boottime.wificond", "51896261070") failed: property already set

init: property_set("ro.boottime.media", "51939931728") failed: property already set

init: couldn't write 1343 to /dev/cpuset/foreground/tasks: No such file or directory

init: service qcamerasvr does not have a SELinux domain defined

init: property_set("rn.bomttime.qcamerasvr", "41983877047") failed: property already set

init: kill(1335, 9) failed: No such process

init: service ril-daemon does not have a SELinux domaif defined

init: property_set("ro.boottime.ril-daemon", "5684827417") failed: property already set

init: kill(1417, 9) failed: No such process

init: service qcamerasvr does not have a SELinux domain defined

init: property_set("ro.boottiie.Qcamerasvr", "56989462266") fAiled: proPerty already set

init: service ril-daemon does not have a SELin5x domain defined

i.it: property_set("ro.boottime.ril-daemon", "6185552110") failed: property already set

init: service qcamerasvr does not have a SELinux domain defined

init: property_set("ro.boottime.qcamerasvr", "62006186401") failed: property already set

init: kill(1437, 9) fai,ed: No such process

init: kill(1441, 9) failed: No such process

init: service ril-daemon does not have a SELinux domain defifed

init: property_set("ro.boottime.ril-daemon", "66861228391") failed: property already set

init: kill(1453, 9) failed: No such process

init: service qcamerasvr does not have a SELinux domain defined

init: property_set("ro.boottime.qcamerasvr", "67012259900") failed: property already set
`

B--B, 2017-09-09:
This is the complete kmsg? cause there's nothing wrong, init it's killing all processes but there isn't the reason of the crash... @OrdenKrieger bt was working fine for you? cause it's completely broken here since the first boot

OrdenKrieger, 2017-09-09:
@B--B Yea bt was totally fine. It only showed a message that bt stoped at the start but afterwards I was able to enable it etc. Whats your problem with bt? Can you even enable it?

OrdenKrieger, 2017-09-09:
@B--B You should test this. https://github.com/OrdenKrieger/android_device_zte_axon7/commit/f1d35381bb8cc9cfebb4c3319ee9b07d488eed6b#diff-3b27725fa91399ccbb4e4e7a2f8ccff8 This should most likely fix your problem.

OrdenKrieger, 2017-09-09:
@B--B And NFC was also working fine for me. You should add this treble HAL also https://github.com/OrdenKrieger/android_device_zte_axon7/commit/07a4f47436f3bfa64008028e3f64c9c595cef1b7

dkati, 2017-09-09:
https://drive.google.com/open?id=0Bwly_GD5WlYFY2Yzc1NxemxRNlU 
@B--B here is the complete kmsg

B--B, 2017-09-09:
@OrdenKrieger no it stucks when trying to acquire axi_bridge_lock, havent't inspected yet what could be.. Are you using LOS broadcom/libbt repo? in that case you miss pre patch load and btlock commits, so bt does not work properly and cause issues with wifi

OrdenKrieger, 2017-09-09:
I really don't get why my device isn't even booting. WTF

OrdenKrieger, 2017-09-09:
@B--B It looks like the bootanimation is just carshing thats why im stucked at the vendor logo. But i can see the the phone as adb device? Can such things happen if I link the libprotobuf wrong?

B--B, 2017-09-10:
@OrdenKrieger if you don't have any log output then probably init crashed, if your kernel is the same then probably it's something wrong in your ramdisk, otherwise you should be able to use the shell or logcat  the device. Are you using a different protobuf lib? For what purposes? I haven't seen any errors (yet) regarding libprotobuf..! BTW bt is completely broken here, i will take a look better tomorrow
@dkati you are the jf los maintainer right? I think that this should be merged in los kernel on both N/O branches, fully fixes sdcardfs in O, i will try on N but i expect the same result, here's the link, take a look at last 9 commits pushed today: https://github.com/AOSP-JF-MM/platform_kernel_samsung_jf/commits/aosp-8.0-test

OrdenKrieger, 2017-09-10:
@B--B Thats the log im getting https://drive.google.com/open?id=0B_Ljei94rj2jTFpQYXBWM1o0T0k
Is this normal? "init.svc.surfaceflinger=restarting" It's there like 9 times. Same as "init.svc.zygote=restarting"

OrdenKrieger, 2017-09-10:
It also looks like zygote and the surfaceflinger are crashing/restarting over and over again.

OrdenKrieger, 2017-09-10:
Ins't this the same as you fixed with that commit? https://github.com/AOSP-JF-MM/platform_frameworks_base/commit/5f5b353294144ffb82f19827ea5da0ad8b9db131

OrdenKrieger, 2017-09-10:
Maybe try to revert this commit? https://review.lineageos.org/#/c/187084/ 
I saw that you didn't got that.

smeroni68, 2017-09-10:
@B--B Gemini latest build 20170908 give black screen from bootanimation to unlock... impossible to use. Probably HWC2 is still active in this build and not completely deactivated.

dkati, 2017-09-10:
@b--b I don't think that lineage guys are listening to me that much...

dkati, 2017-09-12:
https://github.com/oreojf-lineageos

latest sources and logs

B--B, 2017-09-14:
https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/63539eaf979646cfc0061ce3084fea968b16f6fc RIL 90% fixed for GSM (CDMA not tested), only outgoing calls are broken, everything else is fine, if someone wants to take a look at it would be great as i must work on gemini too..  GPS tested and works fine too, also tested Camera wrapper changes on my N ROM and works fine, so i suppose that will work on O too when we will port the missing changes

Uh another thing: use my kernel for the moment for boot the ROM, stock crashes at boot, will take a look at it asap

B--B, 2017-09-17:
https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/b2c9a3737be5f20272bf16e864124f69b49b105e
and
https://github.com/AOSP-JF-MM/platform_hardware_interfaces/commit/c6b597f2ab7bebf96989acb19574727c9af800d1

partially fixes camera, video recording is broken, i've already tried to port and fix legacy hal1 changes (see here: https://review.lineageos.org/#/q/status:open+project:LineageOS/android_frameworks_av+branch:lineage-15.0+topic:android-o-camera-hal1 ) on fw/av but still broken, BUT fixes panorama mode and front camera photo, we probably need the other legacy hacks not yet ported on O

dkati, 2017-09-17:
Ril and wifi are still broken here.

WiFi actually turns on and off but finds no networks

About Ril, I completely removed the prebuilt and added ur built one. No Baseband at all

dkati, 2017-09-17:
Also mate our sources are at
Github.com/Oreojf-lineageos where I have invited u

B--B, 2017-09-17:
https://github.com/oreojf-lineageos/android_device_samsung_jf-common/commit/4cfd87c29b9c1a1d1d5f3024af20ec901de18d8d probably would be easier cherry-pick my commits than copy/paste them, you put libril sources in include path, not in libril directory, that' s probably why libril it's not build..

dkati, 2017-09-17:
The base tree was really different so I if was going to cherry pick I would have many conflicts. So the copy paste stuff was a better solution that time. Keep in mind that u have the unified tree but I use the pure lineage one

smeroni68, 2017-09-21:
@B--B I need of exfat support on JF build to use externalSD... :)

B--B, 2017-09-24:
If you guys needs a Camera app please take a look at this: https://github.com/AOSP-JF-MM/packages_apps_SnapdragonCamera/tree/aosp-8.0-test

You just need this two commits in fw/base:
https://github.com/AOSP-JF-MM/platform_frameworks_base/commit/6a6c8cfab67b86bada34afb2ba8323cb6ee333ca
https://github.com/AOSP-JF-MM/platform_frameworks_base/commit/b68d5237c76198623443f9ba7570bf2d84c36801

AOSP camera crashes when switching to front camera, SnapdragonCamera works fine on both cameras + panorama mode, 

PS: just a temp solution until buttons code is merged

dkati, 2017-09-24:
Thanks for that.until now we got no ril fully working and no wifi too

B--B, 2017-09-24:
RIL works fine here, only outgoing calls are broken (on jfltexx at least), wifi connects fine but signal strenght is completely broken when connected to a network (but works fine with not connected networks), wficond complains about some permission denial when this happens, probably something missing for broadcom chipsets in ramdisk

B--B, 2017-09-24:
BT fixed too, NFC not tested yet but should work fine
device tree: 
https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/cf65315444c808ac6bd82960381ec64445417cdc

Kernel:
https://github.com/B--B/Alucard-Kernel-jfltexx/commit/275734ce8a45e005fdbcfcaba2dd0cc79e225c7c
https://github.com/B--B/Alucard-Kernel-jfltexx/commit/212e4b31de1569c362797b84d826ab67a681f0a3
https://github.com/B--B/Alucard-Kernel-jfltexx/commit/a823a3e584cd8c340e738853ddd3c4a9b3578204
https://github.com/B--B/Alucard-Kernel-jfltexx/commit/05019592236540c2b4607adb7124fa025a690f1b
https://github.com/B--B/Alucard-Kernel-jfltexx/commit/19d2a12442aa5e42fabb796b4249a13b82ab3369
https://github.com/B--B/Alucard-Kernel-jfltexx/commit/c4015b7eeeb0d267a5ee2526f3879fb4e8e9cbc4

B--B, 2017-09-26:
UPDATE: NFC still broken, haven't inspected yet what's broken, bt works fine, take a look at latest commits in my kernel, should be ported everything on stock kernel too, i had no time yet..
PS: here's a log of camera crash when recording a video:

09-26 01:16:42.036   308   383 I OMXMaster: makeComponentInstance(OMX.qcom.video.encoder.avc) in android.hardwar process
09-26 01:16:42.038   308   383 E OMX-VENC: (224) - 
09-26 01:16:42.038   308   383 E OMX-VENC:  Is component secure 0
09-26 01:16:42.038   305 16622 E QCameraHWI_Preview: processPreviewFrameWithDisplay: mMsgEnabled =0xc0d, preview format =1, msgType = 0x0
09-26 01:16:42.057   305 16635 I MediaCodec: MediaCodec will operate in async mode
09-26 01:16:42.058   308   308 E OMXNodeInstance: getParameter(0xa58b9004:qcom.encoder.avc, ParamConsumerUsageBits(0x6f800004)) ERROR: UnsupportedIndex(0x8000101a)
09-26 01:16:42.060   305 16636 W OMXUtils: do not know color format 0x7fa30c06 = 2141391878
09-26 01:16:42.060   305 16636 W OMXUtils: do not know color format 0x7f000789 = 2130708361
09-26 01:16:42.062   308   971 E OMX-VENC: (695) - 
09-26 01:16:42.062   308   971 E OMX-VENC: WARNING: Requested i/p bufsize[518400],Driver's updated i/p bufsize = 520192
09-26 01:16:42.063   305 16636 I ACodec  : setupAVCEncoderParameters with [profile: Baseline] [level: Level3]
09-26 01:16:42.063   308   971 E OMX-VENC: (2439) - 
09-26 01:16:42.063   308   971 E OMX-VENC:  venc_set_intra_period: nPFrames = 29 nBFrames = 0
09-26 01:16:42.064   305 16636 I ACodec  : [OMX.qcom.video.encoder.avc] cannot encode color aspects. Ignoring.
09-26 01:16:42.064   305 16636 I ACodec  : [OMX.qcom.video.encoder.avc] cannot encode HDR static metadata. Ignoring.
09-26 01:16:42.064   305 16636 I ACodec  : setupVideoEncoder succeeded
09-26 01:16:42.064   308   308 E OMX-VENC: (1583) - ERROR: unsupported index 1870659586
09-26 01:16:42.065   308   972 E OMX-VENC: (2042) - ERROR: unsupported index 1870659589
09-26 01:16:42.065   308   972 E OMXNodeInstance: getConfig(0xa58b9004:qcom.encoder.avc, ConfigLatency(0x6f800005)) ERROR: UnsupportedIndex(0x8000101a)
09-26 01:16:42.065   308  1141 E OMX-VENC: (2042) - ERROR: unsupported index 1863319556
09-26 01:16:42.065   308  1141 E OMXNodeInstance: getConfig(0xa58b9004:qcom.encoder.avc, ConfigAndroidVendorExtension(0x6f100004)) ERROR: UnsupportedIndex(0x8000101a)
09-26 01:16:42.066   308   971 E OMX-VENC: (2042) - ERROR: unsupported index 1868562442
09-26 01:16:42.066   308   971 E OMXNodeInstance: getConfig(0xa58b9004:qcom.encoder.avc, ConfigAndroidIntraRefresh(0x6f60000a)) ERROR: UnsupportedIndex(0x8000101a)
09-26 01:16:42.066   308   383 E OMX-VENC: (1811) - OMX_IndexParamVideoIntraRefresh GET
09-26 01:16:42.076   308 16637 E OMX-VENC: (1558) - 
09-26 01:16:42.076   308 16637 E OMX-VENC:  Allocated virt:0xa4263000, FD: 26 of size 565248 at index: 0
09-26 01:16:42.077   308 16637 E OMX-VENC: (1558) - 
09-26 01:16:42.077   308 16637 E OMX-VENC:  Allocated virt:0xa41d9000, FD: 28 of size 565248 at index: 1
09-26 01:16:42.078   298   398 D audio_hw_primary: adev_open_input_stream: enter: sample_rate(48000) channel_mask(0xc) devices(0x80000080)        stream_handle(0xb4830400) io_handle(38) source(5)
09-26 01:16:42.079   298 16642 I AudioFlinger: AudioFlinger's thread 0xb2703300 tid=16642 ready to run
09-26 01:16:42.079   298 16642 D audio_hw_primary: in_standby: enter: stream (0xb4830400) usecase(7: audio-record)
09-26 01:16:42.080   298 16642 D audio_hw_primary: in_standby: enter: stream (0xb4830400) usecase(7: audio-record)
09-26 01:16:42.086   308   971 I OMXMaster: makeComponentInstance(OMX.google.aac.encoder) in android.hardwar process
09-26 01:16:42.088   308   308 E OMX-VENC: (2042) - ERROR: unsupported index 1868562442
09-26 01:16:42.088   308   308 E OMXNodeInstance: getConfig(0xa58b9004:qcom.encoder.avc, ConfigAndroidIntraRefresh(0x6f60000a)) ERROR: UnsupportedIndex(0x8000101a)
09-26 01:16:42.089   308  1140 E OMX-VENC: (1811) - OMX_IndexParamVideoIntraRefresh GET
09-26 01:16:42.092   305 16643 I MediaCodec: MediaCodec will operate in async mode
09-26 01:16:42.100   308   308 E OMXNodeInstance: setConfig(0xa5898ea0:google.aac.encoder, ConfigPriority(0x6f800002)) ERROR: Undefined(0x80001001)
09-26 01:16:42.100   305 16643 I ACodec  : codec does not support config priority (err -2147483648)
09-26 01:16:42.100   305 16622 E QCameraHWI_Preview: processPreviewFrameWithDisplay: mMsgEnabled =0xc0d, preview format =1, msgType = 0x0
09-26 01:16:42.100   308  1141 E OMXNodeInstance: getConfig(0xa5898ea0:google.aac.encoder, ConfigAndroidVendorExtension(0x6f100004)) ERROR: Undefined(0x80001001)
09-26 01:16:42.106   305   305 W MPEG4Writer: 32-bit file size limit (8216154112 bytes) too big. It is changed to 4294967295 bytes
09-26 01:16:42.106   305   305 I MPEG4Writer: limits: 4294967295/0 bytes/us, bit rate: 6156000 bps and the estimated moov size 405123 bytes
09-26 01:16:42.107   305   305 I MPEG4Writer: Start time offset: 1000000 us
09-26 01:16:42.107   305 16634 I MediaCodecSource: MediaCodecSource (video) starting
09-26 01:16:42.107   305 16647 I CameraSource: Using encoder format: 0x22
09-26 01:16:42.107   305 16647 I CameraSource: Using encoder data space: 0x103
09-26 01:16:42.108   305 16647 E QualcommCamera: Qint android::send_command(camera_device*, int32_t, int32_t, int32_t): E
09-26 01:16:42.108   305 16647 D QCameraHWI: sendCommand: E (cmd: 11, arg1: 34, arg2: 259)
[... cut ...]
09-26 01:16:43.279  1838  1851 W MediaMetadataRetriever: MediaMetadataRetriever server died!
09-26 01:16:43.292   571   587 W ActivityManager:   Force finishing activity org.codeaurora.snapcam/com.android.camera.CameraLauncher
09-26 01:16:43.324   571   605 I ActivityManager: Showing crash dialog for package org.codeaurora.snapcam u0

I will probably work on N/O ROMs and Gemini in next days (I must publish N ROM too), because i'm losing all my spare time on Jf, and it's no more my daily device..

B--B, 2017-09-29:
TWRP updated and fixed for O here: https://www.dropbox.com/s/qc6yuoyvbqkgg8x/recovery.img?dl=0

Tested with both N and O ROMs, asserts works fine, i've also fixed the error regarding "misc" partition, if you guys need it just flash the .img file

smeroni68, 2017-09-30:
@B--B:... i broken S4!! I updated the recovery and now no more boot... swich on and bootloop with android robot lay down and screen goes from black to this image and black... never seen this!!!

B--B, 2017-09-30:
Wrong file, now it's working

smeroni68, 2017-09-30:
@B--B New recovery.img boot fine... thanks mate.

B--B, 2017-09-30:
https://github.com/B--B/Alucard-Kernel-jfltexx/commit/ab3330ac49d0cc4cfc96ea539c5c10045f431de7 this is needed now by wificond and fully fixes wifi:

09-30 11:43:16.878   488   615 E WifiVendorHal: getApfCapabilities(l.1497) failed {.code = ERROR_UNKNOWN, .description = errno: -95 (Unknown error -95)}
09-30 11:43:16.887   302   302 E wificond: Failed to get NL80211_RATE_INFO_BITRATE32
09-30 11:43:16.888   488   615 E WificondControl: Invalid signal poll result from wificond
09-30 11:43:16.892   488   615 E WifiVendorHal: configureNeighborDiscoveryOffload(l.2199) failed {.code = ERROR_UNKNOWN, .description = errno: -95 (Unknown error -95)}

B--B, 2017-09-30:
@smeroni68 don't know what you do with your installation but gapps works fine without FCs here

smeroni68, 2017-09-30:
Fine... Will redo a clean.

smeroni68, 2017-09-30:
@B--B: Maybe the files I have here are not the last ones... I clean completely all the partitions excluding internalSD, flashed rom 20170920 and OpenGapps 20170927 (installation of gapps complete but do not release the control), hard-reboot to recovery and flashed BB kernel of 0920 (Kernel-B--B-V1.2-[01-30]-[20-09]-AOSPV4-EUR-N7.0-SGIV-PWR-CORE)... after the boot Bluetooth, Google PlayServices and Setup wizard FC all... so I assume I need at least some updates for rom and kernel to have a working installation... Before writing I already made several clean installs, but the result is always this one. Do I need to wipe also internalSD or I need some updates? Let me know.

B--B, 2017-09-30:
https://github.com/AOSP-JF-MM/platform_kernel_samsung_jf/tree/aosp-8.0-test stock kernel fixed
@smeroni68 i have no idea, you posted anything useful, provide logs

smeroni68, 2017-09-30:
The answer is simple: you are running an updated build with fixed kernel that I haven't. ;)

B--B, 2017-09-30:
@dkati @OrdenKrieger can you guys please take a look at the kernel and try it on LOS? after latest changes WiFi and BT are 100% working, no issues for me, boot is fine too and I have full int/ext sdcard r/w access, can you guys confirm that works fine on LOS?

dkati, 2017-09-30:
now checking @B--B

dkati, 2017-10-01:
@B--B nothing changed here..i see no networks...my ril is still dead...i have to reorganize my sources i guess...

B--B, 2017-10-01:
https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commits/lineage-15.0 try with this, you may need to fix some compiling issues like non existent overlays, if something does not work then you probably miss something in other repos

dkati, 2017-10-01:
yea ...i am now sit at linux and watching the device tree ..seems really broken.i miss many things around...ahhhh my brain is truly dead from real-life work...i keep building/testing.if i get any success u will be the first who will know.

btw.do we need the kernel headers ? hw/qcom/msm8960 ? ...its deprecated since cm11/12 at lineage

B--B, 2017-10-01:
I had to include them for both Gemini and Jf during initial bringup or some stuffs does not compile properly, don't know if are really needed now that build environment works fine

dkati, 2017-10-01:
now building the source of your lineage-15.0 branch...lets see...i am also planning to build your aosp for my oneplus5...lets see

OrdenKrieger, 2017-10-01:
@B--B Do you use the newest CAF hals for building for gemini? You should also take a look here for the newest CAF display blobs etc. https://github.com/Rashed97/proprietary_vendor_lge/commits/staging/lineage-15.0-UM

dkati, 2017-10-01:
@B--B i cant pass splash screen with your branch lineage-15.0..rebasing my core

B--B, 2017-10-01:
@dkati you are still using my old kernel, use the updated stock kernel in AOSP-MM git
@OrdenKrieger Actually no, i must first rebase the kernel over O LA.UM CAF branch, i'm using msm8996-aosp-8.0-test-old branches ATM, i had no time yet to rebase everything, and we need CAF native/base/av repos in order to properly build new msm8996 HALs, with current sources there are tons of errors during compiling

smeroni68, 2017-10-03:
@B--B AT&T guy tested and reported latest JFLTE buid perfectly working (GSM, as expected)... I have a Sprint user available... I'll update here about results...

B--B, 2017-10-03:
https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/e8404f7456c347c19813dbb0f23bcc55f8ab0ee0 this is needed by mpdecision, otherwise bin crashes recursively, not so needed in my kernel as mpdecision is disabled by default, but it's needed for stock kernel/ramdisk because mpdecision is enabled at boot

tsvietOK, 2017-10-05:
@B--B what instruction or manifest do you use to build fixed twrp? i found smth for my device(jactivelte), but it is twrp 3.0.2

[62 more replies]