AOSP-JF-MM/issue_tracking_aosp #120: [DISCUSSION] GEMINI Oreo testing

smeroni68, 2017-09-10:
This is an Internal discussion to report AOSP Oreo issues during testing, logs and so on.

smeroni68, 2017-09-10:
@ktulu84 @smstiv @B--B Let's use this space for reporting and messages...

smeroni68, 2017-09-11:
@B--B I dirty flashed 0910 over 0905... I get the settings menu labels a bit mixed!! Check here:

https://www.dropbox.com/s/kqt10esj6w6fwo6/Screenshot_20170911-213039.png?dl=0
https://www.dropbox.com/s/qu9n8uatekgy4hc/Screenshot_20170911-213043.png?dl=0

I try to clean settings app data, cache... from twrp too... but this is the result. I need to do a clean??!!

ktulu84, 2017-09-12:
There is still the issue with bluetooth streaming, music come from phone speaker instead of car speakers

smeroni68, 2017-09-12:
@B--B Piero confirmed me the menu settings is ONLY a problem on my installation. No problem.

smeroni68, 2017-09-19:
@B--B https://github.com/akhilnarang/devic...3/commits/oreo
OP3 camera has been fixed... dunno if it can help us... to be checked.

ktulu84, 2017-09-20:
Link is broken :D

smeroni68, 2017-09-20:
Fixed, Thanks mate.